costas-basdekis/MarlinGcodeDocumentation #1: Disabled after install

Shama71, 2020-09-28:
Hi/
Plugin disabled after install.

Shama71, 2020-09-28:
Disable after restart octoprint.

costas-basdekis, 2020-09-28:
Hi, thanks for this.

What Python version are you using? If you go to Settings -> Plugin Manager -> At the bottom of the page you can find a line reading something like `Using pip of "/home/pi/oprint/bin/python2", Version 19.3.1`

Also, did you get an error message?

Shama71, 2020-09-28:
Hi.
No error message.

[image]

costas-basdekis, 2020-09-28:
It might be because you have Python 2.7, and the plugin was originally only for Python 3.

If you can check in Settings -> Plugin Manager -> At the bottom of the page you can find a line reading something like `Using pip of "/home/pi/oprint/bin/python2", Version 19.3.1`

In any case, I've pushed an update supporting Version 2.7, see if updating fixes the error.

cp2004, 2020-09-28:
Hey @costas-basdekis!
You need to tag the release in the Github releases page, in order for automatic update notifications to work. Otherwise it will not be pushed out to existing users.

Reference: https://docs.github.com/en/free-pro-team@latest/github/administering-a-repository/managing-releases-in-a-repository

Made it here after investigating a problem reported by a user on Discord.

*(And you may get more of these issues, since the Python 3 migration has only just started.....)*

Shama71, 2020-09-28:
> It might be because you have Python 2.7, and the plugin was originally only for Python 3.
>
> If you can check in Settings -> Plugin Manager -> At the bottom of the page you can find a line reading [...]

Very good.
Error fixed.
Thank you.

closed 2020-09-28 by Shama71

jneilliii, 2020-09-28:
@costas-basdekis you'll want to start utilizing the github releases here for future changes if necessary. The process is pretty straight-forward. You update your setup.py with new version number and then create a new release in the repo here that matches that updated version number. As it stands now, it seems you do not have any releases as of yet, so you might want to create a baseline release now that matches the current version number.

costas-basdekis, 2020-09-28:
@Shama71 Happy to help

@cp2004  @jneilliii Thank you, I was wondering how did OctoPrint know when something is released.

I've retroactively added the releases for my first and current versions.

Is there anything else I need to do?

jneilliii, 2020-09-28:
That's it. Every new release that comes out you just have to do the same thing. Then OctoPrint's software update checks will see the new release tag in your repo and notify the users there's an update on it's normal checking sequence taking into account local caching times, etc.

costas-basdekis, 2020-09-28:
Thank you, I'll tag every new release.

I was having trouble finding in the plugin documentation (which is quite good) what did I need to do to notify OP about new releases, and I thought I had to update `plugins.octoprint.com`, but that's not necessary, right?

cp2004, 2020-09-28:
No, everything is done through the GitHub releases. Once the user has the plugin, they never have to interact with the plugin repository again about your plugins. The check goes straight from the install in OctoPrint to the GitHub API, which has info about the tagged releases.

It can also be good to write release notes about what's changed, since there is a button for that in the update notification that heads there. (though I'm not sure how many people read them... lol)