costas-basdekis/MarlinGcodeDocumentation #5: Add new Prusa GCode Commands (M601, M602)

martinglass, 2020-09-30:
Manufacturers, such as Prusa, have their own specific commands. Can you add the most popular ones to this Documentation Plug-In?

I'm specifically thinking of the M601, M602 (Pause and Resume Print), which are now used in PrusaSlicer filament change routines.

https://www.help.prusa3d.com/en/article/prusa-specific-g-codes_112173
https://reprap.org/wiki/G-code#M601:_Pause_print
https://reprap.org/wiki/G-code#M602:_Resume_print

costas-basdekis, 2020-09-30:
That could be possible.

The issue is that I need to have some form of structure data that gives the commands documentation. Currently I'm looking at [Marlin's documentation source code](https://github.com/MarlinFirmware/MarlinDocumentation/tree/master/_gcode), which has a very nice structure.

* For RepRap, I could probably do the same, they seem structured, but not that much
* For Prusa, not sure, but I could have a look. Do you know if they have source for their documentation?

martinglass, 2020-09-30:
> That could be possible.
>
> The issue is that I need to have some form of structure data that gives the commands documentation. Currently I'm looking at [Marlin's documentation source [...]

If you ask Prusa, they may be able to give you the source, but all I can find is this link:
https://www.help.prusa3d.com/en/article/prusa-specific-g-codes_112173

costas-basdekis, 2020-10-01:
I'm preparing a version that also picks documentation from RepRap (that picks up M601/M602), and it should be up within a day or two.

Thanks for the suggestion - if only I could rename the name of the plugin 😄

martinglass, 2020-10-01:
Glad my feedback was helpful. 
I'm lovin' your plugin. **Apart from Octopi, it's the only plugin I have installed.** 
The ? (search) command is super useful, especially when combined with "- Collapse All", as it gives a neat short summary of the commands, eg ?SD ?M5
(Happy for you to use this as a testimonial)

martinglass, 2020-10-01:
> I'm preparing a version that also picks documentation from RepRap (that picks up M601/M602), and it should be up within a day or two.
> 
> Thanks for the suggestion - if only I could rename the name of the plugin 😄

Nothing wrong with the plugin name, IMHO

costas-basdekis, 2020-10-01:
Haha thanks for the testimonial and the feedback, appreciate it!

About the name, since I'm now starting to support non-Marlin documentation, it's a mismatched 🤷 

A new version is has just been released, so you should be able to see it soon.

Let me know what you think of it

closed 2020-10-09 by costas-basdekis