costas-basdekis/MarlinGcodeDocumentation #6: Great plugin, just some thoughts

fiechr, 2020-09-30:
The basic idea is great, since you get the documentation right there where you need it! Thank you for providing this. :-)

Some thoughts:

I think it would be better, if the tab or the frame showing the documentation, could be placed below the Terminal buttons (below the "Advanced options..." link).

Also some top and bottom padding between the lines, especially to the buttons, would slightly improve readability.

There should be an option or switch to disable the how to use box.

costas-basdekis, 2020-10-01:
Good point, thanks.

I'll push a version in the next couple of days to add a setting

fiechr, 2020-10-01:
Thank you for considering it!

I really like the way its parsing the current command an adds the parameters one by one.

costas-basdekis, 2020-10-01:
Thanks for the feedback!

I've made a new release, let me know what you think of it. If you have suggestions for more styling, let me know, I'm not the most design-inclined person, but I try.

closed 2020-10-09 by costas-basdekis