costas-basdekis/MarlinGcodeDocumentation #7: Plugin still says it's out of date despite being the latest version

NovaViper, 2020-10-01:
Hi, I just updated the plugin from version 0.9.2 to 0.10.0 but Octoprint still says that the plugin isn't update to date. Using Octoprint version 1.4.2.

[image]

costas-basdekis, 2020-10-01:
Hi, thanks for this.

I'm not sure if this is an issue with this plugin specifically, or an issue with how OctoPrint handles updates.

I'd appreciate if @jneilliii could help here, since I followed the cookie cutter for this plugin.

martinglass, 2020-10-01:
Nothing to do with this plugin specifically. It's taken a while for the update to filter through to the Octoprint Plugin database and then be pushed to users. I think that it's all okay now, although if you've updated already you might still see an update warning, saying that the latest version is now available,  Strange!

fnsign, 2020-10-01:
I think there is a typo or kind of version mismatch. Octoprints plugin site say v0.9.2., the release is named v0.10.0 and the master is 0.10.2? This mismatch of versions causes reboot and access problems to the WebUI on my RPi. For the time beeing I have uninstalled the plugin to regain access to Octoprint again.

costas-basdekis, 2020-10-01:
You have a keen eye @fnsign, thank you

I've pushed a new release v0.10.3 @NovaViper

jneilliii, 2020-10-02:
Yeah, versioning of the plugin is entirely controlled by your github repo here. All you have to do is make sure that when you create your release in github that the setup.py file has the matching version number when you create it, because that's how it will download.

NovaViper, 2020-10-02:
@costas-basdekis That fixed it, thanks!

closed 2020-10-02 by NovaViper