costas-basdekis/MarlinGcodeDocumentation #9: Klipper Support

DroneMang, 2020-11-30:
Hi!

Let me first say this is very useful as it is.  I use Klipper and many of the commands are not recognized, I was just wondering if someday there could be support for Klipper.  Thanks!

DroneMang

costas-basdekis, 2020-12-01:
Thanks for the suggestion @DroneMang.

I though most Klipper commands would be included in the RepRap documentation. Do you know if there is Klipper documentation for those commands somewhere?

cmfrolick, 2020-12-27:
@costas-basdekis 

Klipper G-Code documentation can be found at https://www.klipper3d.org/G-Codes.html

Klipper supports a subset of the RepRap G-Code and has it's own extended command set for managing the firmeware and printer control.

costas-basdekis, 2021-01-17:
I've added support for Klipper now and it'll be available on the next release @cmfrolick

DroneMang, 2021-01-17:
Awesome!

costas-basdekis, 2021-01-22:
It should be live now

closed 2021-01-22 by costas-basdekis