costas-basdekis/MarlinGcodeDocumentation #10: Button icons not present

thelewiss1, 2020-12-09:
- OP 0.15.1
- Python 3.7.3
- Plugin 0.11.0
- Terminal Messaging 0.1.0

[image]

Let ne know if you need anymore information!

costas-basdekis, 2020-12-09:
Hm, very interesting! I'm not adding a new font, so it could be a problem of a font being blocked.

I'll have a look, but can you see if there are any errors in the console/network tab, around fonts?

Thanks for the heads up @thelewiss1 !

thelewiss1, 2020-12-09:
> console/network

[image]
[image]

I'm also getting lots of these errors right after screencapping the console tab: 
[image]
and the terminal kept going blank/using "basic" mode. Not sure what that's about.

costas-basdekis, 2020-12-09:
Hm, no issues there. I'll try to have a look

thelewiss1, 2020-12-18:
> Hm, no issues there. I'll try to have a look

Hello! Any updates? The issue still persists.

costas-basdekis, 2021-01-17:
Hi @thelewiss1 sorry for the delay.

I've had a look in different installations and I couldn't find anything.

2 things to try:
* If you check your network tab, do you see successful entries for `fontawesome-webfont.woff2`?
* If you inspect the missing icon (Ctrl+Shift+C on Windows/Linux and then click on the green square icon), what do you see as the entry for font? Is it `normal normal normal 14px/1 FontAwesome`? If you change it to `14px FontAwesome` does that fix it?

If neither of these work, I'm not sure how to diagnose. Perhaps if you gave me access to your instance?

thelewiss1, 2021-01-17:
> Hi @thelewiss1 sorry for the delay.
>
> I've had a look in different installations and I couldn't find anything.
>
> 2 things to try:
>
> * If you check your network tab, do you see successful entries for [...]

I don't see any network activity for fontawesome*.
This is what I see for your second point:
[image]

I don't have any way for you to access my instance unfortunately. Is there a way i can purge the install of the plugin? I have since installed Octopi on a new Pi and restored from a backup.

costas-basdekis, 2021-01-17:
It would be **very weird** if there is no FontAwesome since it's included with OctoPrint (unless I'm really missing something).

If you click on the `::after` element, you should see `font` in `Styles` sub-tab and replace with `14px FontAwesome`

[image]

BTW the missing icon is something added by the plugin, if it annoys you that's missing you can disable it by going in the plugin settings and unchecking `Add links to sent commands in terminal to get documentation`

You can give access to your instance by either using [Localtunnel](http://localtunnel.github.io/www/) or a remote access OctoPrint plugin

costas-basdekis, 2021-01-17:
Actually, one last thing you can try: replace `font` value with `14px "Font Awesome 5 Brands"`

thelewiss1, 2021-01-17:
I've found what you meant. My mistake.
I've tried 
"normal normal normal 14px "Font Awesome 5 Brands""
"14px "Font Awesome 5 Brands" 
"normal normal normal 14px FontAwesome"
"14px FontAwesome"

to no change.

costas-basdekis, 2021-01-18:
I really not sure how else to help here.

I've just run a fresh install with only MGCD and TM, and it works fine.

I just noticed. Are you saying you have version 0.15.1 of OctoPrint? My version is 1.4.2, and the latest one is 1.5.2.

So if you have such an old installation, I'm afraid there is not that much I can do

thelewiss1, 2021-01-18:
> I really not sure how else to help here.
>
> I've just run a fresh install with only MGCD and TM, and it works fine.
>
> I just noticed. Are you saying you have version 0.15.1 of OctoPrint? My version is [...]

I'm using 1.5.2 of Octoprint. That could have been a typing mistake. I'm printing at the moment, but i will try to remove the plugin again. Is there anything I can delete in files to ensure everything is uninstalled before reinstalling?

costas-basdekis, 2021-01-18:
Ok, haha, 0.15.2 would be difficult to install!

There doesn't seem anything wrong with this plugin, so uninstalling and reinstalling just MGD would probably make no difference.

The problem seems to be with not being able to access a font provided by OP. Unless you use something like LocalTunnel it'd be hard to debug it.

If you're not using this feature best thing would be to disable it.

thelewiss1, 2021-01-21:
> Ok, haha, 0.15.2 would be difficult to install!
>
> There doesn't seem anything wrong with this plugin, so uninstalling and reinstalling just MGD would probably make no difference.
>
> The problem seems [...]

I don't particularly use the feature, so i'll turn it off. I'll just copy and paste the command if I need.

Thanks!

costas-basdekis, 2021-01-21:
Apologies for not figuring a solution.

If you ever figure it out let me know.

closed 2021-02-02 by LewisSpring