costas-basdekis/MarlinGcodeDocumentation #12: Feature request: Add configuration for the documentation repository selections

Sembazuru, 2021-03-13:
Having the check boxes for which documentation repositories are used on the main interface is getting really cluttered. I use consolidated tabs and all the selections don't fit so I can't see all of them:
[image]

It would be nice to move these to a configuration interface to reduce the clutter. Maybe have a quick link to the configuration page (like a pencil icon next to the expand and collapse buttons) instead of always having to find it in the OctoPrint configuration window.

costas-basdekis, 2021-03-15:
Thanks for the suggestion @Sembazuru !

These settings are already present on the settings tab, do you want to also have an option to not show those checkboxes in the console section?

Sembazuru, 2021-03-15:
Silly me. I didn't even look for the configuration and had forgotten it was there. This repository's README.md states "There is no configuration currently." in the configuration section and I was basing at least part of the request on that... https://github.com/costas-basdekis/MarlinGcodeDocumentation#configuration

Yes, I'd like a switch to not show those checkboxes in the console section. This would reduce some of my self-imposed clutter.

costas-basdekis, 2021-03-28:
Now there is an option to hide them from there

closed 2021-03-28 by costas-basdekis