costas-basdekis/MarlinGcodeDocumentation #13: Feature request: Add an option to default to all collapsed.

Sembazuru, 2021-03-13:
Whenever I reload the OctoPrint interface my list of saved g-codes loads with all expanded. I then have to click the `Collapse All` button to see the full list (or more of the full list) like an index. I know, it's just another button press, but I want to be lazy. ;-)

This could be another item to put in the configuration window that I requested in issue #12.

costas-basdekis, 2021-03-15:
Good point, I can consider that.

Sembazuru, 2021-03-15:
Thinking about UI elements, I'm not sure which would be more user intuitive:

- A single checkbox to select collapsed on refresh
- Two radio buttons, O Expanded on refresh ; O Collapsed on refresh
- A two item drop-down list (same options as the radio buttons)

I'd probably go with the two item drop-down list for visual brevity and explicit messaging. But, it's your code and I don't want to be a back-seat coder. ;-)

costas-basdekis, 2021-03-28:
Now there is an option to collapse them on page reload

closed 2021-03-28 by costas-basdekis