costas-basdekis/MarlinGcodeDocumentation #14: Klipper documentation doesn't seem to be working

gcarvalho, 2021-04-05:
I'm starting to use klipper for the first time and the plugin doesn't seem to recognise any of the commands.
I tried switching on plugin logging but the log shows nothing unusual. I also don't seem to find any errors in the inspect console.
Anything I should look into?

costas-basdekis, 2021-04-06:
Thanks for reporting this @gcarvalho!

Can you post some examples of commands that are not recognised? I though I was parsing them properly.

I should probably add some tests

gcarvalho, 2021-04-18:
Hey, apologies, I logged off github and only noticed now.
All of them, at least on my setup. PROBE_CALIBRATE, as an example.

degtyarev708, 2021-07-20:
Standard gcode shows up just fine, however none of klippers "extended gcode" is recognized https://www.klipper3d.org/G-Codes.html

pandel, 2021-10-15:
I think, at least here is something going wrong: https://github.com/costas-basdekis/MarlinGcodeDocumentation/blob/b0eabaf3beb513e27a21c39f6fa03a38c1770943/octoprint_marlingcodedocumentation/parser/parsers/klipper.py#L118

`sibling` is a NavigableString or Tag, I think. If I change the code from line 118 to this

```
        next_text = next(filter(None, (
            sibling.replace('\n', '').strip().strip(':').strip()
            for sibling in list_item.previous_siblings
            if isinstance(sibling, str)
        )), '')
```

then, at least, next_text is not empty. But I don't have enough time to dig into this any further...

closed 2021-11-08 by costas-basdekis