costas-basdekis/MarlinGcodeDocumentation #15: "Include from" causes unsightly overlap

JesseHawkins, 2022-01-28:
I suggest that these changes be made to stop this:

https://i.imgur.com/PHy8UU3.png

Change from:

/octoprint_marlingcodedocumentation/templates/marlingcodedocumentation.jinja2:81:        Include from Marlin
/octoprint_marlingcodedocumentation/templates/marlingcodedocumentation.jinja2:85:        Include from RepRap
/octoprint_marlingcodedocumentation/templates/marlingcodedocumentation.jinja2:89:        Include from Klipper

Change to:

/octoprint_marlingcodedocumentation/templates/marlingcodedocumentation.jinja2:81:        Marlin
/octoprint_marlingcodedocumentation/templates/marlingcodedocumentation.jinja2:85:        RepRap
/octoprint_marlingcodedocumentation/templates/marlingcodedocumentation.jinja2:89:        Klipper

End result:

https://i.imgur.com/DvBgBPH.png

costas-basdekis, 2022-01-28:
Thanks for the suggestion @JesseHawkins 

Do you have a PR you can submit?

Otherwise, I'll try to get it in soon

JesseHawkins, 2022-01-28:
Submitted, however I am using GitHub for the first time in years. I have no clue if I did it correctly.  https://github.com/costas-basdekis/MarlinGcodeDocumentation/pull/16

costas-basdekis, 2022-01-29:
That's great, thanks! Welcome to GitHub

costas-basdekis, 2022-01-29:
It's released, and should be available for update soon

https://github.com/costas-basdekis/MarlinGcodeDocumentation/releases/tag/v0.15.0

Do let me know if you have any ideas or suggestions, and thanks for the PR!

closed 2022-01-29 by costas-basdekis