costas-basdekis/MarlinGcodeDocumentation #17: not an issue but a request

NekkoArms, 2022-02-06:
I am hoping for an option to turn off the "Are you sure you want to replace the current command?" box

costas-basdekis, 2022-02-08:
Thanks for the suggestion @NekkoArms! I'll actually consider completely removing it. It does sound unnecessary.

If you can submit a PR, please do, otherwise I'll try to do it myself soon.

costas-basdekis, 2022-02-09:
@NekkoArms I've removed the prompt (and added a non-intrusive "undo" button).

It should be on OP soon

closed 2022-02-09 by costas-basdekis

NekkoArms, 2022-02-09:
PR?

costas-basdekis, 2022-02-10:
Pull request, meaning you've made some changes in your copy of this code, and have asked me to include those changes in the main repo

For example see https://github.com/costas-basdekis/MarlinGcodeDocumentation/pull/16