costas-basdekis/MarlinGcodeDocumentation #20: [BUG]: Latest v0.17.0 delivers no information on entered gcode

fnsign, 2022-05-12:
Hi,
I installed the latest 0.17.0 release and discovered following bug:
When I enter a gcode (e.g. G28), M-command or ?phrase I receive the error message `Unknown Command`, no matter which source I select.
The attached log file shows, that there is a problem with the plugin.
[octoprint-13.log](https://github.com/costas-basdekis/MarlinGcodeDocumentation/files/8680527/octoprint-13.log)

andrewseago, 2022-05-12:
Same issue here. Tried multiple uninstall and reinstall methods.

Shadow27374, 2022-05-12:
I haven't even tried that yet. But for me, the display has now moved to the bottom and is visible on every page. I personally find this very annoying.

CitizenStile, 2022-05-13:
Also getting the same issue, uninstall/reinstall did not work. However manually installing 0.16 worked as others have said.

valemaio2, 2022-05-14:
I can confirm too that the newest version doesn't display any help for any Marlin commands, and had to manually reinstall version 0.16. If you don't know how to, remove the plugin from the plugin manager (remove all data too when prompted by Octoprint), then go to the spanner settings on the top right, plugin manager, get more, and in the From URL field paste this link:

https://github.com/costas-basdekis/MarlinGcodeDocumentation/archive/refs/tags/v0.16.0.zip

which is basically the zip file for version 0.16.

costas-basdekis, 2022-05-17:
Thanks for the report everyone! Really sorry for the break.

If you open the options for the plug-in, there should be a section about downloading the definitions periodically.

What is the last update date there? What happens if you click on "Update now?"

I'm away, so won't be able to fix this week but will try to reproduce and fix next week

fnsign, 2022-05-17:
If I click the `Update now` or `Reload` nothing happens and ther is also no date for `Last updated`. When I return to the terminalafter clickting the update button, there is still no information available.

The given documentation URL seems to work. When I enter it into my browser then the json file is displayed.

PowerWiesel, 2022-05-17:
there seems to be in issue after fetching the new documentation file:

```
2022-05-17 23:53:08,244 - octoprint.plugins.marlingcodedocumentation - INFO - Updating GCode documentation...
2022-05-17 23:53:08,249 - octoprint.plugins.marlingcodedocumentation - INFO -  * Fetching https://raw.githubusercontent.com/costas-basdekis/gcode-documentation-parser/output/output/all_codes.json
2022-05-17 23:53:08,457 - octoprint.plugins.marlingcodedocumentation - INFO -  * Saving update (719883 bytes)
2022-05-17 23:53:08,460 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin marlingcodedocumentation
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/__init__.py", line 154, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1688, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_marlingcodedocumentation/__init__.py", line 139, in on_api_command
    self.update_documentation()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_marlingcodedocumentation/__init__.py", line 123, in update_documentation
    new_all_codes_path.write_text(contents)
  File "/usr/lib/python3.7/pathlib.py", line 1218, in write_text
    with self.open(mode='w', encoding=encoding, errors=errors) as f:
  File "/usr/lib/python3.7/pathlib.py", line 1186, in open
    opener=self._opener)
  File "/usr/lib/python3.7/pathlib.py", line 1039, in _opener
    return self._accessor.open(self, flags, mode)
PermissionError: [Errno 13] Permission denied: 'new-all_codes.json'
```

costas-basdekis, 2022-05-18:
@PowerWiesel that seems to be exactly the issue! Thank you for the logs.

I'll fix within the weekend. Please downgrade the version for now

costas-basdekis, 2022-05-20:
Hello, I have a release, and I'd like someone that had this issue to test a pre-release version, to verify that everything works correctly.

Would someone be willing to do that?

1. Switch to the virtual env for OctoPrint if you have one
2. Double-check that the plugin is already installed: `pip freeze | grep -i documentation`
3. Install the pre-release: `pip install git+https://github.com/costas-basdekis/MarlinGcodeDocumentation@v0.18.0#egg=Marlin_GCode_Documentation`
4. Restart your server
5. Refresh the page
6. Check the plugin's settings page `Last update status` value
    * If that text is not there, you probably didn't install it properly, or in the wrong virtual env
    * If that text is there but with no value, refresh page, or click on `Update cache` and then refresh the page

If all is good, I'll make it an official release

valemaio2, 2022-05-20:
I've installed the 0.18 version, but not via pip, as support for Python2.7 has been dropped with the latest Octoprint update, and the installation didn't work with pip3; I've installed it manually the same way I've reverted back to 0.16 (see my comment above).

Octoprint shows that I'm running version 0.18 of the plugin (which is expected), and so far it seems to be working, both for explaining single commands and for searching for commands (as in **?home** for example). I will keep testing it as I use it extensively, and will report back if there's any issue.

PowerWiesel, 2022-05-20:
I updated the plugin the way you requested it and the issue seems to be resolved :partying_face: 

Thank you very much :+1: 
[image]
[image]

costas-basdekis, 2022-05-20:
Thank you all, I've just released it, so feel free to update

closed 2022-05-20 by costas-basdekis