costas-basdekis/MarlinGcodeDocumentation #21: 0.17.0 The entire display is below Octoprint

Shadow27374, 2022-05-12:
Since version 0.17.0 the whole display is below Octoprint and no longer in the #term tab where it was before.

[image]

costas-basdekis, 2022-05-17:
Hi, thanks for the report! I'll have a look and see what's the issue.

What's your version of Octoprint?

Shadow27374, 2022-05-17:
That was version 1.7.3 but since today I have 1.8.0. The error also occurs here.

costas-basdekis, 2022-05-20:
Hi @Shadow27374 @gary325, I can't replicate it on my installation or a fresh installation of 1.8.0

Could you please diagnose a bit?

* In Settings -> Plugins -> Marlin GCode Documentation -> Appearance -> Documentation position, if you change it and save, does it make any difference?
* What does the top of the page look like? (scroll to the top, and take a screenshot)
* If you run this snippet in the console (press F12 in Chrome -> Console), what's the ouput? `Array.from($(".octoprint-container .tab-pane")).map(el => el.id)` <- This tells me which tabs are available in the page
* If you run this snippet in the console, what's the output? `[$("#terminal-sendpanel"), $("#term")]` <- This tells me if the elements I'm using to anchor the documentation exist in the page

Any help debugging this would be appreciated

Shadow27374, 2022-05-20:
> * In Settings -> Plugins -> Marlin GCode Documentation -> Appearance -> Documentation position, if you change it and save, does it make any difference?
No difference

> * What does the top of the page look like? (scroll to the top, and take a screenshot)
[image]

> * If you run this snippet in the console (press F12 in Chrome -> Console), what's the ouput? `Array.from($(".octoprint-container .tab-pane")).map(el => el.id)` <- This tells me which tabs are [...]
[image]

costas-basdekis, 2022-05-20:
Thank you @Shadow27374 !

It's exactly what I needed, but it doesn't help, because it's everything I expected there to be :(. I'll put some mitigation in case the issue is that I try to put the elements in too quickly, but otherwise I only have a hunch.

I'll put a potential fix in the next release, and hopefully that will be enough.

costas-basdekis, 2022-05-20:
Hello, I have a release, and I'd like someone that had this issue to test a pre-release version, to verify that everything works correctly.

Would someone be willing to do that?

1. Switch to the virtual env for OctoPrint if you have one
2. Double-check that the plugin is already installed: `pip freeze | grep -i documentation`
3. Install the pre-release: `pip install git+https://github.com/costas-basdekis/MarlinGcodeDocumentation@v0.18.0#egg=Marlin_GCode_Documentation`
4. Restart your server
5. Refresh the page
6. Check the plugin's settings page `Last update status` value
    * If that text is not there, you probably didn't install it properly, or in the wrong virtual env
    * If that text is there but with no value, refresh page, or click on `Update cache` and then refresh the page

If all is good, I'll make it an official release

costas-basdekis, 2022-05-20:
I've just released it, so feel free to update, and let me know if it still persists

closed 2022-05-20 by costas-basdekis

Shadow27374, 2022-05-20:
Just tried it and the error is gone. Everything looks like before and works.

Thanks

gary325, 2022-05-22:
0.18.0 fixed the error described above. Thanks for the quick fix (even if I was of no help).